hammerlab/flowdec #1: No module named 'tfdecon'

vzickus, 2018-09-25:
While running CElegans - Multiple Channel Example.ipynb

Also had a quick look at the source, and could not track tfdecon down, but could be blind.

eric-czech, 2018-09-25:
Ah yep, that should take care of it.  I think some of the notebooks haven't been updated since I picked that name for the repo.

vzickus, 2018-09-26:
By the way, is the project still active? It seems last commit was 6 months ago. If so, do you need any testing done? I think I will be playing around with this, so happy to help (I'm no python expert though!).

eric-czech, 2018-09-26:
Yep for sure, or at least the whole project was something of a precursor to [this project](https://github.com/hammerlab/cytokit) which was for doing some higher throughput analysis on images from a specific immunofluorescence technique, but we haven't ironed all the wrinkles out of the experimental process so publications involving either project are still a WIP.  It's been working well internally though (so I haven't committed any changes in a while) and I'm kind of surprised to see that many stars and forks before any related paper so maybe other people are getting some value out of it.

As far as testing goes, one of the things I was less certain about was how to validate point spread functions -- I'm not sure if that's of particular interest to you but there might be a better way to do that than what I did, which was to generate some with this library and PSFGenerator to see if the degree of similarity was "high enough" (it was very high but I'm not sure at what point it becomes enough).  Or maybe it would be good to have comparisons to some other deconvolution software other than DeconvolutionLab2, either in terms of speed or quality of results?  Another even simpler test could be to simply try it on any 2 or 3D dataset for validation other than these ones: http://bigwww.epfl.ch/deconvolution/index.html#data.  Those are the only commonly referenced examples I'm aware of so really anything outside of that would be helpful.

Anyways, thanks for reaching out and feel free to add any issues you run into -- happy to jump back into development on this if some other people would find it helpful too.

vzickus, 2018-09-27:
Yes, I think this can be a very useful project (I am spreading the word in my previous group part of which did work with wavefront coding microscopy, so might potentially be useful for them). I will have to check out cytokit at some point (perhaps it would be worth putting the link to that project on the landing page?). 

In terms of testing, I would be interested to know how flowdec would work deconvolving images acquired with Airy-beam light-sheet (the system we have is based on: [this article](https://www.nature.com/articles/nmeth.2922). I would also be interested seeing how your code compares to a closed source code provided by the company who supplied the microscope (my worry is that their PSF extracted from real images look a lot worse than the PSFs used in your examples (I assume those are synthetic?)). The only possible issue is that we have an NDA with that company, so I am not sure how much I could share.

P.S.

Speaking of generating PSFs, I wonder if [biobeam](https://github.com/maweigert/biobeam) would be useful for you (I haven't had time to play with this so cannot comment how well it works).

eric-czech, 2018-09-27:
Ah very cool, seems like a great test.  Using biobeam does look like it would be pretty similar to the usual theoretical PSFs used in standard fluroescent imaging (informed by microscope parameters), but  I can't say we've ever tried using an empirical PSF and I'm curious how they even get that without some kind of bead or other known object intentionally embedded in the sample (or imaged separately).

Either way though, can you get a PSF and a raw image/volume from the scope to test as inputs to compare to their outputs?  I'd be surprised if there was much of a difference given that the RL algorithm seems to be the underlying implementation in a couple closed-source systems we probed at too.  As far as I can tell most of the value in proprietary deconvolution software seems to be in blind deconvolution or PSF generation so FWIW my guess would be that as soon as you have a good PSF any open source deconvolution library like this would likely function similarly.

Oh and yea, those synthetic datasets are accompanied by PSFs which I imagine were generated based on some theoretical diffraction model (but likely aren't that different from PSFs used to deconvolve actual data).  Also if it helps, I've heard empirical PSFs with oil immersions, as opposed to air, are sometimes pretty crazy so maybe airy-beam diffraction is similarly inconsistent?  If nothing else it might be interesting to try to deconvolve a raw image with a PSF from biobeam and see how it looks, to avoid any NDA restrictions.

vzickus, 2018-10-01:
Yes, I will aim to get some PSF measurements done in the near future, my only worry is that the motorized stage on one of the light sheet systems we have (LaVision Ultramicroscope II) might be limited to 0.5um steps, which might not be ideal. However, I should be able to obtain some data from my previous lab. Since this is a side project mostly, it might take a long time before I actually get this done (as with all things in academia!).

eric-czech, 2018-10-01:
Ah fair enough, well let me know if I can help with anything at any point.

eric-czech, 2019-03-12:
It's been a while, but @vzickus if you're still working with light-sheet images you might find some recent work from @VolkerH at https://github.com/VolkerH/Lattice_Lightsheet_Deskew_Deconv helpful.  There are great examples in there for building PSFs, running deconvolution (with other libraries too), resampling large volumes w/ GPU-acceleration, etc.

Closing this thread for now but let us know if anything else comes up.

closed 2019-03-12 by eric-czech

vzickus, 2019-03-12:
Apologies @eric-czech that I did not get back anything in the end, just did not find the time to do this initially, and then it slipped my mind!

eric-czech, 2019-03-13:
No worries at all!  I was doing some issue house cleaning and realized that might be relevant for you.