hammerlab/flowdec #5: Add linter and travis build

eric-czech, 2018-10-06:


russellb, 2018-10-07:
I'll poke at this next.

russellb, 2018-10-07:
See the top commits on https://github.com/russellb/flowdec/commits/master for one approach for this.

The first commit creates a tox (https://tox.readthedocs.io/en/latest/) configuration that runs the flake8 utility against the Python source.  It disables all warnings necessary to make it pass successfully.  My earlier patch dropping unused imports resolved one warning pointed out by flake8.  Most of the remaining ones are related to Python formatting conventions (PEP 8).  flake8 can also help find some real issues, like Python 3 compatibility things.

The second commit adds the travis-ci config, which just runs tox.

You can see an example of this config running on this test PR: https://github.com/russellb/flowdec/pull/4

russellb, 2018-10-07:
I'll send it over as a PR if you're happy with it.

eric-czech, 2018-10-08:
Looks good to me!  The couple times I've gone through that before included adding the environments and running tests manually as part of the travis config so thanks for pointing out tox.  I'll try to use that instead from now on.

eric-czech, 2018-10-17:
Looks like this is all set now per https://github.com/hammerlab/flowdec/pull/7 .

Although, what are your thoughts on how to get code analysis as well style enforcement?  I always appreciated the unused imports, misspelled variable names, unreachable code, etc. that comes from pylint so is there something better these days for that?  Is that what you meant by other things flake8 can find?

russellb, 2018-10-29:
Yeah, flake8 reports stuff like that.  I've used flake8 more personally so
that's what i set up here first.  It seems there's some overlap, and some
things each do that the other doesn't.  More complete coverage would come
from just running both, which shouldn't be too hard if you wanted to add
it.  It'd just be some additions to tox.ini, following the flake8 example.

eric-czech, 2018-12-06:
Definitely done now after adding unit tests to tox.ini (https://github.com/hammerlab/flowdec/commit/2421e3577432280b9e1db0c60f2ba6b8de5ccb3a)

closed 2018-12-06 by eric-czech