hammerlab/flowdec #9: Observer function example

VolkerH, 2018-12-06:
Hi, 

first of all thank you for this awesome library. I started experimenting with it a couple of days ago and am getting excellent results. 

Now my issue: I'm trying to use the observer function to write intermediate iteration results to disk (I would like to see how the iterations progress) during deconvolution. There is no documentation on this, so a working example would be greatly appreciated. My own attempts were unsuccessful so far (defining some minimalist observer function).

```
#define an observer function
def observer(*arg):
    print("Observer called")

# Direct deconvolution on skewed llsm dataset
spimrawvol = tifffile.imread("/home/vhil0002/test_vol.tif")
spimskewpsf = tifffile.imread("/home/vhil0002/test_skewed_psf.tif")

algo_skew = tfd_restoration.RichardsonLucyDeconvolver(n_dims=3, start_mode='input', observer_fn=observer).initialize()
aq_skew = fd_data.Acquisition(data=spimrawvol.astype(np.float32), kernel=spimskewpsf.astype(np.float32))
res_skew = algo_skew.run(aq_skew, niter=3)
``` 

I can see that it gets called once and then I get an exception (probably when it returns) with a lengthy traceback. I've tried returning integers (32 and 64 bit) from the observer or just returning all the input arguments but get pretty much the same error no matter what.

```
Observer called

---------------------------------------------------------------------------
InvalidArgumentError                      Traceback (most recent call last)
~/su62_scratch/volker_conda/llsm/lib/python3.6/site-packages/tensorflow/python/client/session.py in _do_call(self, fn, *args)
   1333     try:
-> 1334       return fn(*args)
   1335     except errors.OpError as e:

~/su62_scratch/volker_conda/llsm/lib/python3.6/site-packages/tensorflow/python/client/session.py in _run_fn(feed_dict, fetch_list, target_list, options, run_metadata)
   1318       return self._call_tf_sessionrun(
-> 1319           options, feed_dict, fetch_list, target_list, run_metadata)
   1320 

~/su62_scratch/volker_conda/llsm/lib/python3.6/site-packages/tensorflow/python/client/session.py in _call_tf_sessionrun(self, options, feed_dict, fetch_list, target_list, run_metadata)
   1406         self._session, options, feed_dict, fetch_list, target_list,
-> 1407         run_metadata)
   1408 

InvalidArgumentError: 0-th value returned by pyfunc_6 is int64, but expects int32
	 [[{{node while/observer}} = PyFunc[Tin=[DT_FLOAT, DT_INT32], Tout=[DT_INT32], token="pyfunc_6", _device="/job:localhost/replica:0/task:0/device:CPU:0"](while/Maximum/_85, while/Identity/_87)]]

During handling of the above exception, another exception occurred:

InvalidArgumentError                      Traceback (most recent call last)
<ipython-input-37-007b7300fa24> in <module>
     12 
     13  #   %%timeit
---> 14 res_skew = algo_skew.run(aq_skew, niter=3)

~/su62_scratch/volker_conda/llsm/lib/python3.6/site-packages/flowdec/restoration.py in run(self, acquisition, niter, session_config)
    211     def run(self, acquisition, niter, session_config=None):
    212         input_kwargs = dict(niter=niter, pad_mode=self.pad_mode, pad_min=self.pad_min, start_mode=self.start_mode)
--> 213         res = self._run(acquisition, input_kwargs, session_config=session_config)
    214         return DeconvolutionResult(res['result'], info={k: v for k, v in res.items() if k != 'result'})
    215 

~/su62_scratch/volker_conda/llsm/lib/python3.6/site-packages/flowdec/restoration.py in _run(self, acquisition, input_kwargs, session_config)
     81             data_dict = {self.graph.inputs[k]:v for k, v in acquisition.to_feed_dict().items()}
     82             args_dict = {self.graph.inputs[k]:v for k, v in input_kwargs.items() if v is not None}
---> 83             res = sess.run(self.graph.outputs, feed_dict={**data_dict, **args_dict})
     84             return res
     85 

~/su62_scratch/volker_conda/llsm/lib/python3.6/site-packages/tensorflow/python/client/session.py in run(self, fetches, feed_dict, options, run_metadata)
    927     try:
    928       result = self._run(None, fetches, feed_dict, options_ptr,
[... 95 lines cut ...]
  File "/home/vhil0002/su62_scratch/volker_conda/llsm/lib/python3.6/site-packages/tensorflow/python/ops/control_flow_ops.py", line 2939, in _BuildLoop
    body_result = body(*packed_vars_for_body)
  File "/home/vhil0002/su62_scratch/volker_conda/llsm/lib/python3.6/site-packages/flowdec/restoration.py", line 304, in body
    decon, i = tf_observer([decon, i], self.observer_fn)
  File "/home/vhil0002/su62_scratch/volker_conda/llsm/lib/python3.6/site-packages/flowdec/tf_ops.py", line 31, in tf_observer
    observe_op = tf.py_func(_observe, tensors, tf.int32, stateful=True, name='observer')[0]
  File "/home/vhil0002/su62_scratch/volker_conda/llsm/lib/python3.6/site-packages/tensorflow/python/ops/script_ops.py", line 457, in py_func
    func=func, inp=inp, Tout=Tout, stateful=stateful, eager=False, name=name)
  File "/home/vhil0002/su62_scratch/volker_conda/llsm/lib/python3.6/site-packages/tensorflow/python/ops/script_ops.py", line 281, in _internal_py_func
    input=inp, token=token, Tout=Tout, name=name)
  File "/home/vhil0002/su62_scratch/volker_conda/llsm/lib/python3.6/site-packages/tensorflow/python/ops/gen_script_ops.py", line 129, in py_func
    "PyFunc", input=input, token=token, Tout=Tout, name=name)
  File "/home/vhil0002/su62_scratch/volker_conda/llsm/lib/python3.6/site-packages/tensorflow/python/framework/op_def_library.py", line 787, in _apply_op_helper
    op_def=op_def)
  File "/home/vhil0002/su62_scratch/volker_conda/llsm/lib/python3.6/site-packages/tensorflow/python/util/deprecation.py", line 488, in new_func
    return func(*args, **kwargs)
  File "/home/vhil0002/su62_scratch/volker_conda/llsm/lib/python3.6/site-packages/tensorflow/python/framework/ops.py", line 3274, in create_op
    op_def=op_def)
  File "/home/vhil0002/su62_scratch/volker_conda/llsm/lib/python3.6/site-packages/tensorflow/python/framework/ops.py", line 1770, in __init__
    self._traceback = tf_stack.extract_stack()

InvalidArgumentError (see above for traceback): 0-th value returned by pyfunc_6 is int64, but expects int32
	 [[node while/observer (defined at /home/vhil0002/su62_scratch/volker_conda/llsm/lib/python3.6/site-packages/flowdec/tf_ops.py:31)  = PyFunc[Tin=[DT_FLOAT, DT_INT32], Tout=[DT_INT32], token="pyfunc_6", _device="/job:localhost/replica:0/task:0/device:CPU:0"](while/Maximum/_85, while/Identity/_87)]]

```

eric-czech, 2018-12-06:
Hey @VolkerH ,

This should be all set now.  Thanks for outlining the problem well and to avoid this for any future users, I added a related [Example Notebook](https://github.com/hammerlab/flowdec/blob/3b7fdb9d2cd29b62e7e97142353ec76408dc44b5/python/examples/notebooks/Monitoring%20Iterations.ipynb), a unit test for observer functions, and updated the doc string to explain what the signature should be.

Here's a short self-contained example:

```python
from flowdec import data as fd_data
from flowdec import restoration as fd_restoration
imgs = []

def observer(img, i, *args):
    # img: float32 array containing current state of deconvolved image
    # i: 1-based iteration counter
    imgs.append(img)
    if i % 10 == 0:
        print('Observing iteration', i)

acq = fd_data.bars_25pct()
algo = fd_restoration.RichardsonLucyDeconvolver(n_dims=3, observer_fn=observer)
algo.initialize().run(acq, niter=100)
assert len(imgs) == 100
```

I'll close this out for now but let me know if you run into issues with that again.

closed 2018-12-06 by eric-czech

VolkerH, 2018-12-07:
Thanks a lot for the quick fix @eric-czech 
I won't get to try before Tuesday due to other commitments. Will report back.

VolkerH, 2018-12-11:
Thanks again, just tested it. Works for me.

eric-czech, 2018-12-13:
Excellent --  feel free to post any other issues or feature requests you might have @VolkerH and thanks for the update!