hammerlab/flowdec #14: Installing flowdec[gpu] on Windows10 causes "Win10: ImportError: DLL load failed: The specified module could not be found"

bryantChhun, 2019-03-11:
installing both tensorflow and flowdec through the pip install instructions results in the above problem.

This is a tensorflow problem and can be solved by rolling back the CUDA install to v9.0, and getting the appropriate cudnn.  See here: https://github.com/tensorflow/tensorflow/issues/22794

It's also possible that I simply didn't have the right cudnn, as this library is not included in TF by default.

chalkie666, 2020-04-22:
@eczech  as of today, 22 april 2020 this below is what worked for me.
Maybe you could add in the readme what versions of CUDA, and tensor flow you are using and know work. Because the command 
pip install flowdec[tf_gpu]
gave me a system that was broken using newest available stuff:
CUDA 10.2, cudnn10.2,
and from the pip install command above I got tensorfow2.1 
Maybe this combination of cuda and tf isn't supposed to work for flowdec yet? 

cheers
Dan

Here's what worked for me today as copied out of my test script:
# things were installed in this order and the script works and used the GPU
#cuda toolkit 10.0 
#cudnn-10.0 7.6.34.38 installed into C:/tools/
pip install flowdec
     not pip install flowdec[tf_gpu]   
omitting the tf_gpu option, so it leaves tensorflow-gpu uninstalled, because by default
 by now it installs v2.1 of tensorflow which doesnt seem to work for flowdec? 
 pip install tensorflow-gpu==1.14.0 (2.0 might work...? maybe needs higher cuda version)
 Need windows env variables pointing to cuda stuff:  CUDA and CUPTI and another related library cuDNN
 SET PATH=C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.0\bin;%PATH%
 SET PATH=C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.0\extras\CUPTI\libx64;%PATH%
 SET PATH=C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.0\include;%PATH%
 SET PATH=C:\tools\cuda\bin;%PATH%

eric-czech, 2020-04-24:
Hey @bryantChhun, we haven't tested anything on TF 2.x and the version bound was set at > 1.x a while back before that release so I'm not surprised it doesn't work.  I'll fix that at some point but definitely try with TF 1.14.x instead.

joaomamede, 2020-09-05:
I had some trouble in windows only. The linux machines are fine by calling pip install with flowdec[tf_gpu]

For windows, I needed to install the specific version of tensorflow-gpu and then flowdec[tf_gpu] with pip first, then remove tensorflow from the pip installation, then install tensorflow-gpu (my case 10.1) with anaconda.

eric-czech, 2020-09-08:
Thanks @joaomamede.  At some point we should probably update the docs to mention that it isn't worth using the pip install for this beyond experimentation since none of the MKL/BLAS/ATLAS extensions are likely to be configured correctly without a conda install anyhow.

pawlowac, 2020-09-17:
Hi @eric-czech 
Following up on the installation question. What is the best method for installation now?

eric-czech, 2020-09-21:
Hi @pawlowac, conda is preferable.  E.g. `conda install --file flowdec/python/requirements.txt` and then pip install flowdec in the same environment