hammerlab/flowdec #22: TypeError: unexpected keyword argument: img_num

naureeng, 2019-05-03:
I am using a virtual conda environment with TensorFlow 1.13.1 and I cannot import the C. elegans dataset. This is a picture of the error I get
[image]

eric-czech, 2019-05-05:
As a workaround, you could switch to scikit-image 0.14.0 (i.e. ```pip install scikit-image==0.14.0```).

I'll leave this open though in case someone (probably me) gets a chance to either either pin it as a requirement or figure out a way to make it work with newer scikit-image versions.

naureeng, 2019-05-07:
Thanks for the reply! There is a bug between numpy 1.16 and scikit-image 0.14.0/0.14.1
https://github.com/scikit-image/scikit-image/issues/3586

I installed scikit-image 0.14.2 to get it working

closed 2019-05-13 by naureeng