hammerlab/flowdec #23: Command Line Deconvolution Troubleshooting

naureeng, 2019-05-13:
I have a flowdec virtual environment set up on our Cluster and look forward to using it! I am having trouble setting up deconvolution with the command line:

(1) the examples folder for the command line interface is missing 
--data-path=flowdec/datasets/bars-25pct/data.tif \
--psf-path=flowdec/datasets/bars-25pct/kernel.tif \
echo '{"na": 0.75, "wavelength": 0.425, "size_z": 32, "size_x": 64, "size_y": 64}' > /tmp/psf.json

None of these paths exist

(2) I have multi-channel images similar to the C. elegans example. My file is:
2048 x 2048 x 54 (2 color channels with 27 slices each) 
I get:
  File "/nfs/nhome/live/naureeng/.conda/envs/flowdec/lib/python3.7/site-packages/flowdec/data.py", line 24, in __init__
    raise ValueError('Number of data and kernel dimensions must be 1, 2, or 3')
ValueError: Number of data and kernel dimensions must be 1, 2, or 3

The PSF file I have was done with a bead and is:
100 x 100 x 100

(3) Can we use the fd_data function to load our own datasets?

Thanks!

eric-czech, 2019-05-13:
Hi @naureeng , to your points:

1) I would imagine that the files being loaded are somehow coming in as either 4 or 5 dimensional.  Or at least that error can only happen if ```.ndim``` of the loaded array is greater than 3.  To verify that, you could try something like:

```python
from flowdec import data as fd_data
from flowdec import restoration as fd_restoration
img, psf = # ... however you'd like to load the image arrays, perhaps skimage.io.imread
assert img.ndim == 3 # should be (2048 x 2048 x 54)
assert psf.ndim == 3  # should be (100 x 100 x 100)
algo = fd_restoration.RichardsonLucyDeconvolver(3, pad_mode='none').initialize()
res = algo.run(fd_data.Acquisition(img, psf), niter=25).data # Run deconvolution
```

2) I think the above example might answer your question.  You just need to pass 2 3D image arrays to ```fd_data.Acquisition```.  If you're asking about how to load an image stored as separate files though, you might find this idiom useful:  

```python
files = glob.glob('/some/path/*.tif')
img = np.stack([ skimage.io.imread(f) for f in files ])
```

If you have multi-channel images, then the ```img``` array above will probably have 4 dimensions instead of 3 in which case you could deconvolve them by iterating over them.  For example:

```python
# Assuming img and psf have shape (x, y, z, channels):
res = np.stack([ algo.run(fd_data.Acquisition(img[..., i], psf[..., i])).data for i in range(img.shape[-1])
```

Hope that helps, but I don't imagine anything in fd_data would be very helpful for loading images since they're probably just following the patterns in one of those examples above.

naureeng, 2019-05-13:
Awesome, thank you so much for your quick reply @eczech! I really appreciate it. I got your code snippet working but I get this error for padding:

tensorflow.python.framework.errors_impl.InvalidArgumentError: Paddings must be non-negative: -37 -37
         [[{{node Pad}}]]
         [[{{node result}}]]

Any ideas?
Thanks!

eric-czech, 2019-05-13:
Could you post the code you used (one of my snippets has pad_mode='none' and I'm not sure which you're talking about) and the results of ```.shape``` for the images you're loading?  Hard to say otherwise.

naureeng, 2019-05-13:
[image]

eric-czech, 2019-05-13:
Ah right, the image extent has to be greater than or equal to the psf extent in all dimensions.  There's supposed to be a better error for that (will get fixed in https://github.com/hammerlab/flowdec/issues/24).  

In other words, it's ok to use a PSF that's a lot smaller than the image but not the other way around.  I'd suggest padding the image with zeros out to 100 z slices or better yet, cropping the PSF or regenerating it with different dimensions if it was theoretical, assuming that 27 z-planes is still enough to fit the PSF.

naureeng, 2019-05-14:
Awesome 👍 I got memory errors when I used a smaller PSF with less z-slices and solved the issue by making a PSF image scaled to be identical to the target image (2048 x 2048).

closed 2019-05-14 by naureeng