hammerlab/flowdec #25: ResourceExhaustedError

naureeng, 2019-05-29:
Dear Eric,

I tried to deconvolve two identical stacks. One works and the other results in a ResourceExhuastedError. I've attached the ImageJ headers:

(1) 2048x2048pixels  (does not work)
(2) 204.4x204.4microns (works)

[image]

eric-czech, 2019-05-29:
Hi @naureeng, it looks like the one is 2044x2044 pixels and the other is 2048x2048 right?  They should certainly both work if the larger one works so my very first question would be whether or not you're doing this in two separate jupyter notebooks (or IPython kernels).  If you are, then whichever one you ran first will allocate all (or most) of the GPU memory and the other will likely fail.  You'd have to run them in the same process or set the TensorFlow options that control how much memory is used to make it less than 50% for each process.

If it's not that, then if you send some code or more information on what you're doing I could probably help.

naureeng, 2019-05-31:
I'm not sure what it was but the micron units works a lot faster for some reason... I will close the issue for now.

closed 2019-05-31 by naureeng