hammerlab/flowdec #26: ResourceExhausedError for z-stack

naureeng, 2019-06-12:
Dear Eric,
I've been able to do batch processing with your help! I now have z-stacks that are:
512 x 512 x 348

I am running into a memory issue:
```
ResourceExhaustedError (see above for traceback): OOM when allocating tensor with shape[512,1024,1024] and type complex64 on /job:localhost/replica:0/task:0/device:GPU:0 by allocator GPU_0_bfc
	 [[node while/FFT3D (defined at /nfs/nhome/live/naureeng/.conda/envs/flowdec/lib/python3.7/site-packages/flowdec/restoration.py:289) ]]
Hint: If you want to see a list of allocated tensors when OOM happens, add report_tensor_allocations_upon_oom to RunOptions for current allocation info.

	 [[node result (defined at /nfs/nhome/live/naureeng/.conda/envs/flowdec/lib/python3.7/site-packages/flowdec/restoration.py:318) ]]
Hint: If you want to see a list of allocated tensors when OOM happens, add report_tensor_allocations_upon_oom to RunOptions for current allocation info.

```

However, I'm allocating an entire node (600GB). Is there a better way to get around this? dask? 
Many thanks for your time!

eric-czech, 2019-06-12:
How memory do you have in the GPU?  Using ```pad_mode='none'``` when creating the RichardsonLucyDeconvolver instance will use less memory at the risk of boundary artifacts (which won't matter if you don't have objects near the border).  Otherwise 512x1024x1024 should still fit on your GPU unless it has <= 4G, so you may have multiple notebooks/sessions running.

naureeng, 2019-06-17:
I ended up splitting the stack and deconvolving it successfully. Thanks for the reply! I have 600G memory on the GPU. It could have been due to multiple jobs.

closed 2019-06-17 by naureeng