hammerlab/flowdec #32: Estimation of required memory

piby2, 2020-06-07:
Hello.
I am trying to deconvolve a 101x2160x2560 (DxWxH) image with a 3x9x9 (DxWxH) PSF. I have tf-gpu 1.14.0 installed, and have 11GB card. 

```
import tensorflow as tf
from flowdec import data as fd_data
from flowdec import restoration as fd_restoration
os.environ['TF_CPP_MIN_LOG_LEVEL'] = '3'
os.environ['CUDA_VISIBLE_DEVICES'] = '1'
from skimage.external import tifffile as tif
from skimage.io import imsave, imread

config = tf.ConfigProto()
config.gpu_options.allow_growth = True
sess = tf.Session(config=config)

imname = 'new.tif'
actual = imread(imname)
actual = np.asarray(actual, dtype=np.float32)
print(actual.shape)

kname = 'psf1.tif'
kernel = imread(kname)
kernel = np.asarray(kernel,dtype=np.float32)
kernel = np.transpose(kernel,(2,1,0))
print(kernel.shape)

t1 = time()
print('Initializing')
algo = fd_restoration.RichardsonLucyDeconvolver(actual.ndim, pad_mode='none').initialize()
print('Running')
res = algo.run(fd_data.Acquisition(data=actual, kernel=kernel), niter=15).data
res = np.asarray(res,dtype=np.uint16)

t2 = time()
print(t2-t1)
print(res.shape)
imsave('decon_result.tif', res)

```
I get segmentation fault. However with a smaller (101x1000x1000) image, the same PSF works with approximately 8815 MB gpu memory.

I have tried both with allow_growth = True and allow_growth = False. In both cases, I got segfault.

This is the segfault message with 101x2160x2560 image. 
Can you please tell me how to estimate GPU memory based on image size and PSF size, so that I can efficiently do chunking?

```
[roys5@mh02112259dt tmp]$ python flowdec_example.py
WARNING:tensorflow:From flowdec_example.py:20: The name tf.ConfigProto is deprecated. Please use tf.compat.v1.ConfigProto instead.

WARNING:tensorflow:From flowdec_example.py:22: The name tf.Session is deprecated. Please use tf.compat.v1.Session instead.

2020-06-07 17:48:13.460340: I tensorflow/core/platform/cpu_feature_guard.cc:142] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
2020-06-07 17:48:13.468494: I tensorflow/stream_executor/platform/default/dso_loader.cc:42] Successfully opened dynamic library libcuda.so.1
2020-06-07 17:48:13.685218: I tensorflow/compiler/xla/service/service.cc:168] XLA service 0x5585f3cc4d80 executing computations on platform CUDA. Devices:
2020-06-07 17:48:13.685291: I tensorflow/compiler/xla/service/service.cc:175]   StreamExecutor device (0): GeForce RTX 2080 Ti, Compute Capability 7.5
2020-06-07 17:48:13.690803: I tensorflow/core/platform/profile_utils/cpu_utils.cc:94] CPU Frequency: 2294660000 Hz
2020-06-07 17:48:13.696609: I tensorflow/compiler/xla/service/service.cc:168] XLA service 0x5585f3f2ed20 executing computations on platform Host. Devices:
2020-06-07 17:48:13.696668: I tensorflow/compiler/xla/service/service.cc:175]   StreamExecutor device (0): <undefined>, <undefined>
2020-06-07 17:48:13.698584: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1640] Found device 0 with properties:
name: GeForce RTX 2080 Ti major: 7 minor: 5 memoryClockRate(GHz): 1.545
pciBusID: 0000:04:00.0
2020-06-07 17:48:13.699119: I tensorflow/stream_executor/platform/default/dso_loader.cc:42] Successfully opened dynamic library libcudart.so.10.0
2020-06-07 17:48:13.702031: I tensorflow/stream_executor/platform/default/dso_loader.cc:42] Successfully opened dynamic library libcublas.so.10.0
2020-06-07 17:48:13.704691: I tensorflow/stream_executor/platform/default/dso_loader.cc:42] Successfully opened dynamic library libcufft.so.10.0
2020-06-07 17:48:13.705260: I tensorflow/stream_executor/platform/default/dso_loader.cc:42] Successfully opened dynamic library libcurand.so.10.0
2020-06-07 17:48:13.708751: I tensorflow/stream_executor/platform/default/dso_loader.cc:42] Successfully opened dynamic library libcusolver.so.10.0
2020-06-07 17:48:13.711392: I tensorflow/stream_executor/platform/default/dso_loader.cc:42] Successfully opened dynamic library libcusparse.so.10.0
2020-06-07 17:48:13.718264: I tensorflow/stream_executor/platform/default/dso_loader.cc:42] Successfully opened dynamic library libcudnn.so.7
[... 143 lines cut ...]
2020-06-07 17:48:59.802383: I tensorflow/core/common_runtime/bfc_allocator.cc:800] InUse at 0x7fdb59010100 next 9 of size 256
2020-06-07 17:48:59.802405: I tensorflow/core/common_runtime/bfc_allocator.cc:800] InUse at 0x7fdb59010200 next 10 of size 256
2020-06-07 17:48:59.802427: I tensorflow/core/common_runtime/bfc_allocator.cc:800] InUse at 0x7fdb59010300 next 11 of size 1280
2020-06-07 17:48:59.802449: I tensorflow/core/common_runtime/bfc_allocator.cc:800] Free  at 0x7fdb59010800 next 15 of size 512
2020-06-07 17:48:59.802471: I tensorflow/core/common_runtime/bfc_allocator.cc:800] InUse at 0x7fdb59010a00 next 16 of size 1024
2020-06-07 17:48:59.802494: I tensorflow/core/common_runtime/bfc_allocator.cc:800] Free  at 0x7fdb59010e00 next 18446744073709551615 of size 979456
2020-06-07 17:48:59.802520: I tensorflow/core/common_runtime/bfc_allocator.cc:809]      Summary of in-use Chunks by size:
2020-06-07 17:48:59.802546: I tensorflow/core/common_runtime/bfc_allocator.cc:812] 9 Chunks of size 256 totalling 2.2KiB
2020-06-07 17:48:59.802571: I tensorflow/core/common_runtime/bfc_allocator.cc:812] 1 Chunks of size 1024 totalling 1.0KiB
2020-06-07 17:48:59.802595: I tensorflow/core/common_runtime/bfc_allocator.cc:812] 1 Chunks of size 1280 totalling 1.2KiB
2020-06-07 17:48:59.802620: I tensorflow/core/common_runtime/bfc_allocator.cc:812] 1 Chunks of size 64000 totalling 62.5KiB
2020-06-07 17:48:59.802644: I tensorflow/core/common_runtime/bfc_allocator.cc:812] 2 Chunks of size 2233958400 totalling 4.16GiB
2020-06-07 17:48:59.802668: I tensorflow/core/common_runtime/bfc_allocator.cc:816] Sum Total of in-use chunks: 4.16GiB
2020-06-07 17:48:59.802696: I tensorflow/core/common_runtime/bfc_allocator.cc:818] total_region_allocated_bytes_: 8590983168 memory_limit_: 10562958132 available bytes: 1971974964 curr_region_allocation_bytes_: 8589934592
2020-06-07 17:48:59.802725: I tensorflow/core/common_runtime/bfc_allocator.cc:824] Stats:
Limit:                 10562958132
InUse:                  4467985408
MaxInUse:               4467985408
NumAllocs:                      25
MaxAllocSize:           2233958400

2020-06-07 17:48:59.802801: W tensorflow/core/common_runtime/bfc_allocator.cc:319] ***************************______________________***************************_______________________*
2020-06-07 17:48:59.802853: W tensorflow/core/framework/op_kernel.cc:1502] OP_REQUIRES failed at pad_op.cc:137 : Resource exhausted: OOM when allocating tensor with shape[101,2160,2560] and type float on /job:localhost/replica:0/task:0/device:GPU:0 by allocator GPU_0_bfc
Segmentation fault (core dumped)
```

eric-czech, 2020-06-08:
At a minimum, the amount of memory needed would be:

101*2160*2560 * (4 bytes for float32) * 3 (PSF, image, and intermediate results) = ~6.7G

The worst part of that in your case is certainly that the PSF is padded out to the size of the image.  There are deconvolution implementations out there that work on different dimensions for each, but there was no clear way to do that with the tf.fft functions.  Although, I've never seen a disparity between the two be even close to as large before -- how'd you end up with a 3x9x9 PSF?  That seems curiously small (is this a 2x widefield image or something?).

Anyways, in my own experiments I would always estimate about 2x on top of that minimum estimate (I don't know the TF internals enough to say why) so I'm not at all surprised that it doesn't fit in 8G.

piby2, 2020-06-08:
Thank you. The PSF is small because it is for a Lightsheet image with 5x5x5um resolution.
Also is the first dimension number of slices? io.imread makes tiff images to numpy array with DxWxH. Is that the correct orientation (DxWxH) to feed the numpy array to fd_data?

eric-czech, 2020-06-08:
You sure you're not getting HxW from imread (instead of WxH)?  DxHxW is what users typically do, but the RL algorithm technically doesn't care.

piby2, 2020-06-08:
> You sure you're not getting HxW from imread (instead of WxH)? DxHxW is what users typically do, but the RL algorithm technically doesn't care.

Sorry, you are right, it DxHxW. I want to make sure D is the first axis.

eric-czech, 2020-06-08:
Ok cool, DxHxW is good then.

piby2, 2020-06-15:
> At a minimum, the amount of memory needed would be:
>
> 101_2160_2560 * (4 bytes for float32) * 3 (PSF, image, and intermediate results) = ~6.7G
>
> The worst part of that in your case is certainly that [...]

After lots of experiments, I realized that the total memory required will be 8 times the float32 memory of one input + some overhead. The input (and psf) is always padded to next power of 2.

So for an image of 1468x1663x17, the required memory was 2048x2048x32x4x8/(1024**3) = 4GB + some overhead. In this scenario, the overhead was 2453MB, so total memory (as shown in nvidia-smi) was 4096+2453 = 6549 MB

For an image of size 1819x2063x17, the required memory was 2048x4096x32x4x8/(1024**3) = 8GB + some overhead. The overhead was 2453MB, so total memory (as shown in nvidia-smi) was 8192+2453 = 10645MB

For an image of size 844x953x17, the required memory was 1024x1024x32x4x8/(1024**3) = 1GB + some overhead. The overhead was 917MB, so total memory (as shown in nvidia-smi) was 1024+917 = 1941MB

I did not have a good idea of how to compute the overhead, but over various image sizes, the required memory computation was consistent. Hopefully, this will help optimize chunking large images to utilize the full GPU memory.

Specs: tf-gpu 1.14.0, python 3.6.8

closed 2020-06-15 by piby2