hammerlab/flowdec #34: Comparison with scikit-image / cupyimg

chrisroat, 2020-07-04:
I was starting to benchmark flowdec against this [implemention](https://github.com/mritools/cupyimg/blob/82ab473b1480afc3f7652d6d14277024988c5658/cupyimg/skimage/restoration/deconvolution.py#L356) in the [cupyimg](https://github.com/mritools/cupyimg) package, which aims to port ndimage and scikit-image onto the GPU using cupy.

I noticed that the scikit-image (and hence cupyimg's port) implementation suffer from problems with values near 0.   I think it because it sometimes leads to a divide-by-zero.  This explains why all the toy example code injects noise on the demo images!

Is flowdec a bit immune to this problem because of this [filtering step](https://github.com/hammerlab/flowdec/blob/f565324b09a1fbd0c8b676fe7a160f07a6d35d79/python/flowdec/restoration.py#L301) or [this clipping](https://github.com/hammerlab/flowdec/blob/f565324b09a1fbd0c8b676fe7a160f07a6d35d79/python/flowdec/restoration.py#L306)?

As a concrete example, the astronaut example you use to demonstrate edge ringing will not work directly in scikit-image's implementation without the added Poisson noise.  Though, I've found it does reasonably well if you just rescale+shift the image from 0->1 to 0.5->1.

[My reason for trying cupyimg is that I'm already using cupy in my docker image, and adding tensorflow's 1.5GB is making my images even larger.  For now, at least, I'm sticking with flowdec because it works on our data, since we hit the divide-by-zeros with the naive implementation.]

eric-czech, 2020-07-05:
Hey @chrisroat, yea if I recall correctly the RL likelihood model is invalid if any of the intermediate arrays contain negative values (being based on poisson distributions).  I saw both of those things (the filtering + clipping) in other implementations so I added them after some initial errors thrown from TF fft functions.

It's probably worth filing an issue on cupyimg given that it's more actively maintained.  There is definitely precedent for that kind of adjustment in imagej and deconvolutionlab2 so I don't imagine they wouldn't want to incorporate it.

chrisroat, 2020-07-06:
The cupyimg people are just mimicking [scikit-image](https://github.com/scikit-image/scikit-image/blob/772d7be825c38dd9bf994f929fc573c50be8db00/skimage/restoration/deconvolution.py#L369), so I may have to start there.  Thanks for the info.

closed 2020-07-06 by chrisroat