hammerlab/flowdec #40: How to release the memory after flowdec processing

joaomamede, 2020-11-19:
The only thing I could find was this but it worked:

```
from numba import cuda
cuda.select_device(0)
cuda.close()
#the memory was released here!
cuda.select_device(0)
```

closed 2020-11-19 by joaomamede

eric-czech, 2020-11-19:
Thanks for sharing!  Several people have asked about that in the past and I don't think anyone has thought about (or at least I haven't) trying to use a different library to free GPU memory held by the same Python process.  I'm surprised that works but good to know there is at least one way to force it.

joaomamede, 2020-11-19:
I think tensorflow 1 had a function but not TF2.x.

Happy to help, and thank you for flowdec, I have an accepted article and another submitted with flowdec deconvolved data, it really helped having this!