hammerlab/flowdec #43: Inpractical GPU memory usage

SebastienTs, 2022-07-19:
While indeed extremely fast the GPU memory required for deconvolution is just not practical.

For a reasonably small 38 MB image stack (1024x1024x19, 16-bit) and a tiny PSF (32x32x16), almost 8 GB GPU is required. 
As such, I could only process a slightly larger image (1396x1396x19, 16-bit) on a RTX 3090 (24 GB).

I don't know if it is a Tensorflow / Windows 10 issue but something is probably not right as for FFT computation, even assuming that the images are casted to 32-bit floating point and that the PSF is padded to image size, I would not expect more than say a 8x ratio between image size and memory requirement.

eric-czech, 2022-07-21:
Hey @SebastienTs, there are a number of reasons the memory usage is often way more than you might expect:

1. The PSF is padded to the size of the image, so it doesn't matter if it's smaller
2. Tensorflow FFT operations don't support sub 32 bit types (or at least they didn't when this was written)
3. The image array is copied in intermediate states
4. Other tensorflow memory overhead (e.g. as observed in https://github.com/hammerlab/flowdec/issues/32)
5. Often most importantly, the default padding mode pushes all dimensions up to next power of 2 (so in your case 1024x1024x19 would become 1024x1024x32 for both the image and the PSF).

I would suggest you try `pad_mode='2357'` which is a more memory-efficient but less computationally-efficient (sometimes) method added in https://github.com/hammerlab/flowdec/pull/18.

Apart from that, the only other practical option is to chunk the arrays as in [Tile-by-tile deconvolution using dask.ipynb](https://github.com/hammerlab/flowdec/blob/master/python/examples/notebooks/Tile-by-tile%20deconvolution%20using%20dask.ipynb).

SebastienTs, 2022-07-21:
Thanks a lot for your reply! I had 1, 2 and 5 quite clear in mind but even then do you really believe that 3 and 4 could explain the remaining 30x memory overhead (from 270 MB to 8 GB)?

If that is the case I can sleep peacefuly but it sounds like a real lot to me and I want to make sure that something is not misconfigured or extremely suboptimal for the Tensorflow version I am using...

I would happily consider the cucim alternative that is recommened but unfortunately my code needs to run on a Windows box.

eric-czech, 2022-07-21:
Hm well 10x wouldn't surprise me too much but 30x does seem extreme.  When it comes to potential TF issues I really have no idea.

You should take a look at this too if you haven't seen it: https://github.com/hammerlab/flowdec/issues/42#issuecomment-1113233702.  Some of those alternatives to this library may be Windows friendly.

joaomamede, 2022-07-26:
Have a look in my repo:

https://github.com/joaomamede/mamedelab_scripts/blob/main/notebooks/Google2021_Deconvolve_Live_gui.ipynb

I basically use dask to divide the images and assemble them again when the GPU mem is not enough.

This is the bioformats version (older, might have some tweaks to be done)
https://github.com/joaomamede/mamedelab_scripts/blob/main/notebooks/2021deconvolve_live_bioformats.ipynb

They should be able to run on google collaboratory version if you'd like to tweak around.

You also need the libraries at:
https://github.com/joaomamede/mamedelab_scripts/blob/main/notebooks/libraries/deco_libraries.py

hope it helps